KingAbad/PEM_AutonomousCart #19: 4G module aansturen met STM

ozi-e, 2020-10-09:


meto51, 2020-10-22:
nog aan het wachten voor een simkaart, ik denk dat het via m-center moet gebeuren de configuratie van de 4g-module en niet echt programmeren, tenzij je extra features erop wilt zoals automatisch sms'en versturen. dit kan ik pas uitzoeken als ik de simkaart ook echt heb.

meto51, 2020-10-24:
https://github.com/Seeed-Studio/ublox_lara_r2_pi_hat

meto51, 2020-10-24:
ik heb verder gezocht voor python code die alleen raspberry pi ondersteunt, maar het is in principe dezelfde chip en python code die via linux geinstalleerd moet worden. ik zal dit ook proberen of het gaat lukken.

meto51, 2020-10-24:
dus het zal via uart verbonden moeten worden op de jetson nano en zo proberen het aan de praat te krijgen

meto51, 2020-10-29:
https://eu-api.asm.skype.com/v1/objects/0-weu-d3-8c3b06a36514ec9044890c8d4dee377c/views/imgo

meto51, 2020-10-29:
[image]

meto51, 2020-10-29:
dit is een voorbeeld van hoe het op de jetson nano eruit ziet in ieder geval bij mij orange, de jetson nano leest de module zonder enige library, driver, enz. Maar het is niet kiesbaar en je kan het niet verbinden. Ik heb verder nog op de raspberry pi geprobeerd maar geeft het zelfde probleem, als je via commandline de devices gaat bekijken is het serieel wel verbonden maar je kan het niet verbinden helaas. Verder heb ik nog opgezocht en van alles uitgeprobeerd via cmd en andere instellingen maar ook helaas niet aan de praat gekregen.

meto51, 2020-10-29:
verder dacht ik dat de probleem aan de jetson nano was omdat het niet geupdate kon worden via commando : sudo apt-get update, hij geeft een error aan. Aangeraden om de Jetson nano eens te flashen en de drivers opnieuw installeren maar helaas heb ik daar geen tijd voor gehad.

meto51, 2020-10-29:
[image]

meto51, 2020-10-29:
[image]

meto51, 2020-10-29:
[image]

meto51, 2020-10-29:
via cmd de disconnected proberen aan te zetten maar het lukte helaas niet.

meto51, 2020-10-29:
[image]

meto51, 2020-10-29:
volgens mikroe zou het zo moeten werken...

meto51, 2020-10-29:
er waren drivers en installs beschikbaar, maar alleen voor andere hats en shields(raspberry pi) die dezelfde lara-chip hadden. 
Veel code en voorbeelden voor de mikroe development board zelf via microbasic software die 300 euro is.

meto51, 2020-11-02:
ik stuur nu via de 4g module deze messagem het  werkt...

closed 2020-11-02 by meto51